qiweimao/ESP32-Datalogger #13: ESP-NOW Network Management

NomiBobby, 2024-05-13:
What kind of table does the gateway maintain?

How does gateway and nodes establish connection?

Use JSON or other formats for communication?

NomiBobby, 2024-05-16:
- send request from nodes will fail, how to handle?
- gateway power cycle will lose all mac addresses and channel setting, should i add this info to nvs?

NomiBobby, 2024-05-17:
The following functionalities still need to be developed:
- master should be able to send all logger control API to nodes
- nodes should be able to transfer readings to master
 - trunking

NomiBobby, 2024-05-17:
Interesting read on ESP-NOW

NomiBobby, 2024-05-23:
Interesting read on LoRa struct:
https://github.com/sandeepmistry/arduino-LoRa/issues/49

qiweimao, 2024-05-26:
API:

1. Logger System configuration: implemented as a form.
2. Data collection configuration.

qiweimao, 2024-05-28:
Now the problem is, data collection configuration might exceed 250 bytes, which we should avoid. Explore option to compress and decompress this.

For most sensor readings, 250 bytes will be more than enough. SAA might be a issue due to large number of nodes.

qiweimao, 2024-07-19:
implemented

closed 2024-07-19 by qiweimao